acmerobotics/road-runner-quickstart #332: .turnTo() has unexpected stutter

madacker, 2023-12-12:
### RR FTC Version

0.1.8

### Observed Behavior

Using MeepMeep version 0.1.2:

```
            myBot.runAction(myBot.getDrive().actionBuilder(new Pose2d(12, 63, Math.toRadians(-90)))
                            .lineToY(60)
                            .turnTo(Math.toRadians(180))
                            .build());
```
The robot turns 90 degrees and then immediately the robot jumps back in orientation and again does the turn. The stutter depends on the angle: 181 degrees doesn't exhibit the problem in this example while 45 also does. The stutter manifests in both MeepMeep and on the field.

rbrott, 2023-12-12:
I believe this should be fixed in`1.0.0-beta6`

closed 2023-12-12 by rbrott

madacker, 2023-12-13:
Verified the fix on the robot using the repro and a couple of other quick sanity tests.