acmerobotics/road-runner-quickstart #333: I am getting an hardware map is outdated error when I init my drive.

ayoda1122, 2023-12-13:
### RR FTC Version

0.1.9

### Observed Behavior

I have my declaration in my auto base that is extended to my opmodes

rbrott, 2023-12-16:
Can you post the exact error message? Do you mean that your Lynx firmware is outdated?

ayoda1122, 2023-12-17:
Yes, that was the error. But at the moment I cant run it to get the error again

rbrott, 2023-12-17:
Okay then you need to follow the instructions here: https://ftc-docs.firstinspires.org/en/latest/ftc_sdk/updating/hub_firmware/Updating-Hub-Firmware.html

ayoda1122, 2023-12-17:
Sorry just found the error message I was thinking of another error here is the current error
[image]

rbrott, 2023-12-17:
You're constructing the drive object too early. Wait to construct it until `runOpMode()`/`init()`.

closed 2023-12-17 by rbrott