acmerobotics/road-runner-quickstart #334: Template drive action code.

alexDHS0, 2023-12-14:
I don't know if our repository is missing samples or something but we don't actually have any examples of how to build a drive trajectory and run that trajectory using 1.0 actions instead of the old 0.5 system. It's making it very hard to figure things out.

c24al2, 2023-12-15:
The quickstart does not ship with samples. There's a good sample [here](https://gist.github.com/TheOutcastVirus/fc0de0afdcb37808288904e308a67dc7), by The Outcast Virus.

rbrott, 2023-12-16:
There is also some documentation here: https://rr.brott.dev/docs/v1-0/actions/. If you have an op mode from 0.5, I'd be happy to show how to adapt it to fit the new system.

rbrott, 2025-01-12:
Late, but there's also https://rr.brott.dev/docs/v1-0/guides/centerstage-auto/ now

closed 2025-01-12 by rbrott