acmerobotics/road-runner-quickstart #337: Custom Spline :00000 V1.0

IKKNIGHT, 2023-12-18:
Need Custom Spline rn rn rn V1.0

IKKNIGHT, 2023-12-18:
[image]
Reality

[image]
Expectation

rbrott, 2023-12-18:
You can drop down to the low-level spline API to have more control over derivatives:
```java
        QuinticSpline1d xSpline = new QuinticSpline1d(
                new DualNum<>(new double[]{ x_0, dx_0, d2x_0 }),
                new DualNum<>(new double[]{ x_1, dx_1, d2x_1 })
        );
        QuinticSpline1d ySpline = new QuinticSpline1d(
                new DualNum<>(new double[]{ y_0, dy_0, d2y_0 }),
                new DualNum<>(new double[]{ y_1, dy_1, d2y_1 })
        );
        QuinticSpline2d spline = new QuinticSpline2d(xSpline, ySpline);
```
A `QuinticSpline2d` in Road Runner world is more like a quintic spline segment that just goes between two knots.

closed 2024-12-14 by IKKNIGHT

Overtest42, 2024-12-22:
Sorry to reopen this old issue, but if you supposedly were able to generate a Quintic2dSpline, how could you then use that spline to create into a trajectory?

rbrott, 2024-12-22:
> Or how could you use existing TrajectoryBuilder methods to create the same path as a custom Quintic Spline?

Generally you should stick to what's provided by the builders, and you should usually be able to achieve the paths you desire through that interface. 

> but if you supposedly were able to generate a Quintic2dSpline, how could you then use that spline to create a trajectory?

It's certainly possible to turn manually constructed splines into a trajectory. Here are the steps:

* Wrap the spline in an `ArclengthReparamCurve2d` to turn it from a `PositionPath<Internal>` to a `PositionPath<Arclength>`. (Mathematically this converts the spline representation from **x**(t) where 0 <= t <= 1 to **x**(s) where 0 <= s <= spline length and ||**x**'(s)|| = 1.)
* Wrap the position path in `TangentPath` to get a `PosePath` which includes the robot heading along the path. (There are also different choices of heading "paths"/interpolators here that can be used with `HeadingPosePath`.)
* Generate a displacement motion profile for the path using `profile()` and use it to construct a `DisplacementTrajectory`. (That displacement trajectory can also be used to construct a `TimeTrajectory` if desired.)

In other words, it's a bit involved. The API is composable to a fault, allowing for almost maximal flexibility at the cost of ease of use. It's mainly designed to be consumed by builders / easier path creation abstractions like those provided elsewhere in Road Runner (and those classes are decent references for how to slot all of the objects together).