acmerobotics/road-runner-quickstart #338: path follower go vroom vroom v1.0

IKKNIGHT, 2023-12-20:
I was told to make this so ok

rbrott, 2023-12-21:
I sketched an approach here: https://rr.brott.dev/docs/v1-0/guides/path-following/. 

Not all of the details are filled in, and I would only recommend this to advanced teams.

ProNoob1029, 2024-02-24:
Hi, I'm trying to follow the path following guide (pun semi-intented), but I'm having a hard time understanding the exercise to the reader part:

> "try taking the derivative of the orthogonality constraint between the path point/derivatives and the query point"

Could the guide maybe be expanded with a more in-depth explanation and more code examples? I'm really interested in implementing path following even if it takes a more custom approach. I'm also a bit confused about the last part about:

> "Then take the raw Pose2dDual<Arclength> from before and compute the maximum path displacement velocity/time derivative that satisfies the constraints"

What I managed to understand from the explanation until now is written bellow:

```kotlin
val pathDisplacement = project(timeTrajectory.path, pose.position, lastDisplacement)

val targetPoseArc = timeTrajectory.path[pathDisplacement, 3]

val timeDisplacement = timeTrajectory.profile.dispProfile[pathDisplacement]

val targetPos = targetPoseArc.reparam(timeDisplacement)
```

rbrott, 2024-02-25:
> Could the guide maybe be expanded with a more in-depth explanation and more code examples? 

Perhaps in the future. I do like it in its current form as an invitation to experimentation instead of dogmatically prescribing best practices. The trajectory following system that people usually associate Road Runner with is the ordained approach and comes with the benefit of elaborated instructions.

> "try taking the derivative of the orthogonality constraint between the path point/derivatives and the query point"

This is a hint toward a possible enhancement to simpler path followers. It's not something I would recommend starting with. I would focus on getting the most basic path follower running on your robot and observing how it runs. In fact, I would probably start with a pure feedback controller that chooses a pose setpoint by projecting the pose estimate onto the path and adding a fixed lookahead distance. Start playing around with that and go from there.

> "Then take the raw Pose2dDual from before and compute the maximum path displacement velocity/time derivative that satisfies the constraints"

This is also gesturing at another enhancement that you don't need (and shouldn't have!) to start.

closed 2024-12-14 by IKKNIGHT