acmerobotics/road-runner-quickstart #343: Error with RR 1.x and a tank drive.

PythonAtSea, 2023-12-29:
### RR FTC Version

0.1.11

### Observed Behavior

So I tried to set up Roadrunner 1.x. Our robot is a tank drive, and we're using drive encoders. Following the guide [here](https://rr.brott.dev/docs/v1-0/tuning/), everything seemed to be going well. However, when we got to the ManualFeedbackTuner, we got this error: https://drive.google.com/file/d/1ljwnY8K-ViJXSkme4i6236-PnQLuoGep/view?usp=sharing . I tried again with the sppline test, and I got the same error.

rbrott, 2023-12-30:
Seems like I didn't get around to adding array support in the flight recorder system. You can comment out that line for now:
```diff
diff --git a/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/TankDrive.java b/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/TankDrive.java
index 04b7d88..137b986 100644
--- a/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/TankDrive.java
+++ b/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/TankDrive.java
@@ -186,8 +186,8 @@ public final class TankDrive {
             meanRightPos /= rightEncs.size();
             meanRightVel /= rightEncs.size();
 
-            FlightRecorder.write("TANK_ENCODERS",
-                    new TankEncodersMessage(leftReadings, rightReadings));
+//            FlightRecorder.write("TANK_ENCODERS",
+//                    new TankEncodersMessage(leftReadings, rightReadings));
 
             TankKinematics.WheelIncrements<Time> twist = new TankKinematics.WheelIncrements<>(
                     new DualNum<Time>(new double[] {
```

rbrott, 2024-02-19:
Should be fixed now

closed 2024-02-19 by rbrott