acmerobotics/road-runner-quickstart #347: Documentation Issues/Common Problems

j5155, 2024-01-12:
In the FTC Discord we see many teams with issues caused by missing things in the documentation. 
#### Dead wheel hardware config:
Many people see this line: 
> The code expects the two parallel encoders to be named "par0" and "par1" and the perpendicular one to be named "perp".

and assume it means that they need to dedicate a slot on the hub to the encoder, and it should be named as such in configuration. Instead, this line could be something like this:

> When using odometry, drivetrain encoders are not necessary and the odometry modules can be connected to the drivetrain encoder slots instead. Change the names in ThreeDeadWheelLocalizer to reflect the motor name of the slot the modules were connected to. (Note that, [for technical reasons](https://blog.eeshwark.com/robotblog/rev-hub-quadrature), ports 0 and 3 are more accurate and should be used for the parallel odometry pods.

#### Dead wheel positioning
Tuning, or at least estimating, the dead wheel positions is a requirement for a successful tune. However, the current documentation's instructions to do so are easy to miss, especially when skimming. Because of this, many teams do not realize that they should tune the odometry wheel positions here, and report strange, difficult to debug issues caused by this (#270 #291).

Potentially a yellow warning, like the one about requiring a kS and kV to tune this properly, would help a lot here? Or putting odometry before wheel, and putting it in bold. 

These issues may technically  be the fault of teams, but I think they would be seen a lot less if it was made even more obvious in the documentation what the right thing to do is.

rbrott, 2024-01-12:
Thanks for the suggestions. I published these changes to the docs:

```diff
@@ -10,7 +10,7 @@ involves running several op modes and adjusting a set of parameters that capture
 the behavior of the bot. The entire affair shouldn't take more than a few hours,
 and it can go much faster once you know the ropes.
 
-Read this page very carefully, especially if it's your first time. Most details
+**Read this page very carefully**, especially if it's your first time. Most details
 are only mentioned once and you don't want to miss them. Failing to fully
 complete one step in the sequence will often affect the next step, and you may
 be forced to start over when you don't get satisfactory results at the end.
@@ -95,6 +95,18 @@ localizers:
   two parallel encoders to be named `"par0"` and `"par1"` and the perpendicular
   one to be named `"perp"`.
 
+{{< hint >}}
+If you choose to use dead wheels, drive encoders are not needed, and those
+encoder ports can be used for dead wheel encoders. You can change the names in
+`TwoDeadWheelLocalizer`/`ThreeDeadWheelLocalizer` to reflect the motor names of
+the slots the encoders were connected to.
+
+Note that, for [technical
+reasons](https://blog.eeshwark.com/robotblog/rev-hub-quadrature), ports 0 and 3
+on REV hubs are more accurate at high speeds and should be used for the parallel
+dead wheels.
+{{< /hint >}}
+
 {{< hint >}}
 Check out
 [gm0](https://gm0.org/en/latest/docs/common-mechanisms/dead-wheels.html) for
@@ -296,8 +308,8 @@ Go to `http://192.168.43.1:8080/tuning/dead-wheel-angular-ramp.html` and click
 the "Latest" button. Copy the `kS` and `kV` values from `ForwardRampLogger` into
 the appropriate boxes and click the "update" button. Use the instructions from the
 `ForwardRampRegression` analysis to fill in `trackWidthTicks` in your
-drive class and the wheel position fields in your dead wheel class. There should
-be multiple plots. Scroll down to make sure you don't miss any.
+drive class and the wheel position fields in your dead wheel class. **There should
+be multiple plots. Scroll down to make sure you don't miss any.**
 
 {{< hint warning >}}
 To get an accurate `trackWidthTicks` value, you need accurate values for `kS`,
```

Vector5233, 2024-01-26:
So… Team Vector would potentially be interested in putting in some hours helping out with a tuning guide for 1.x that would build on the docs you have already.   Would you be open to that?

rbrott, 2024-01-26:
Sure, I'm happy to accept suggestions for the docs. I do like to retain editorial control of what's actually added though. I would suggest starting with smaller suggestions to get an idea of how the process works before investing time in a giant piece of writing that might not fit my personal style.

closed 2024-02-12 by j5155