acmerobotics/road-runner-quickstart #349: NavX IMU: Localization is off.

SAK-20744, 2024-01-18:
getting correct values from the NavX, but the localization is jerky. Not sure if it is because of an incorrect parallel odo pod placement, or navX heading velocity. (Manually calculated, as the NavX does not give heading velocity normally.) In this clip, the robot is just spinning in a circle.

https://github.com/acmerobotics/road-runner-quickstart/assets/121530772/f5cbad89-0274-4ffb-ba1e-d65cc5d7a5b6

I believe it has something to do with the way I calculated external heading, but I am not 100% sure.

  @Override
    public Double getExternalHeadingVelocity() {
        double velInitial = Math.toRadians(-navx_device.getYaw());
        double timeInitial = System.nanoTime();
        double velFinal = Math.toRadians(-navx_device.getYaw());
        double timeFinal = System.nanoTime();

        double deltaV = velFinal - velInitial;
        double deltaT = timeFinal - timeInitial;
        double headingVel = deltaV/deltaT;

        return headingVel;
    }

rbrott, 2024-01-23:
Heading velocity does not affect localization. Looks like your parallel odo pod positions are off.

SAK-20744, 2024-01-23:
thats a relief. Thanks.

closed 2025-01-12 by rbrott