acmerobotics/road-runner-quickstart #353: Combining Commands in RR (FTCLib)

SharkDjokovic, 2024-01-22:
I'm trying to run a RR command in an OpMode by initializing it in init, and it keeps stalling in init and then says "Stuck in Stop." This only happens when I add multiple splines.

When I add the trajectories in start, the init process works fine, and then it stalls in a position when I click start, waits 15 seconds, and then says "Stuck in Stop." Everything works fine when I remove the trajectories.

OpMode:
```java
package org.firstinspires.ftc.teamcode.Drive.OpModes.TestAutonomous;

import com.acmerobotics.dashboard.config.Config;
import com.acmerobotics.roadrunner.geometry.Pose2d;
import com.acmerobotics.roadrunner.geometry.Vector2d;
import com.arcrobotics.ftclib.command.CommandScheduler;
import com.arcrobotics.ftclib.command.InstantCommand;
import com.arcrobotics.ftclib.command.ParallelCommandGroup;
import com.arcrobotics.ftclib.command.SequentialCommandGroup;
import com.arcrobotics.ftclib.command.WaitCommand;
import com.qualcomm.robotcore.eventloop.opmode.Autonomous;
import com.qualcomm.robotcore.eventloop.opmode.OpMode;
import com.qualcomm.robotcore.util.ElapsedTime;

import org.firstinspires.ftc.robotcore.external.hardware.camera.WebcamName;

import org.firstinspires.ftc.teamcode.Drive.DriveConstants;
import org.firstinspires.ftc.teamcode.Drive.SampleMecanumDrive;
import org.firstinspires.ftc.teamcode.TrajectorySequences.TrajectorySequence;
import org.firstinspires.ftc.teamcode.TrajectorySequences.TrajectorySequenceBuilder;
import org.firstinspires.ftc.teamcode.Utility.CommandBase.Commands.DriveCommand;
import org.firstinspires.ftc.teamcode.Utility.CommandBase.Commands.RestCommand;
import org.firstinspires.ftc.teamcode.Utility.CommandBase.Commands.SpecialOuttakeCommand;
import org.firstinspires.ftc.teamcode.Utility.CommandBase.Commands.TapeDropCommand;
import org.firstinspires.ftc.teamcode.Utility.Hardware.RobotHardware;
import org.firstinspires.ftc.teamcode.Utility.Vision.RedRightProcessor;
import org.firstinspires.ftc.vision.VisionPortal;
import org.opencv.core.Scalar;

@Autonomous
@Config

public class NEWRedRight extends OpMode {
    private VisionPortal visionPortal;
    private RedRightProcessor colorMassDetectionProcessor;

    private RobotHardware robot;
    private ElapsedTime time_since_start;
    private double loop;

    public TrajectorySequence movement1Left;

    @Override
    public void init() {
        CommandScheduler.getInstance().reset();
        robot = new RobotHardware(hardwareMap);

        CommandScheduler.getInstance().registerSubsystem(robot.armSystem);
        CommandScheduler.getInstance().registerSubsystem(robot.claw);
        CommandScheduler.getInstance().registerSubsystem(robot.angleOfArm);
        CommandScheduler.getInstance().registerSubsystem(robot.driveSubsystem);

        telemetry.addData("Not Ready: ", "Not able to proceed to camera detection... Restart robot now.");
        telemetry.update();

        robot.claw.grabBoth();

        Scalar lower = new Scalar(0, 110, 65);
        Scalar upper = new Scalar(8, 200, 150);
        double minArea = 100;

        colorMassDetectionProcessor = new RedRightProcessor(
                lower,
                upper,
[... 211 lines cut ...]
        return drive.isBusy();
    }

    public void turn(double radians) {
        drive.turnAsync(radians);
    }

    public List < Double > getWheelVelocities() {
        return drive.getWheelVelocities();
    }

    public void stop() {
        drive(0, 0, 0);
    }

    public Pose2d getPoseVelocity() {
        return drive.getPoseVelocity();
    }

    public Localizer getLocalizer() {
        return drive.getLocalizer();
    }

}
```

rbrott, 2024-01-23:
Maybe the splines are just taking too long to generate? Ideally you could time that section and print out how long it takes. The easiest knob to make generating splines faster is reducing the [resolution](https://github.com/acmerobotics/road-runner-quickstart/blob/41a45054885fd4c7bc11e5cc6fff52bd7596d339/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/trajectorysequence/TrajectorySequenceBuilder.java#L33) (or at some point consider migrating to RR 1.0 which generates trajectories more quickly).

SharkDjokovic, 2024-01-23:
What do you suggest reducing the resolution to? Will this still be accurate? What does the resolution change?

rbrott, 2024-01-23:
You can change the resolution to something like 2-4 inches. The resolution affects the sampling of constraints for motion profile generation. The higher the number is, the coarser the profile is and the less strictly constraints will be followed.

SharkDjokovic, 2024-01-24:
You mentioned constraints, are these constraints like velocity or specific to accuracy? Will decreasing the resolution cause motions to be less accurate?

SharkDjokovic, 2024-01-24:
Also, if I switch to rr1.0, my arm subsystem uses RR0.5.6's MotionProfile class. Would I need to copy in the Kotlin function for that?

rbrott, 2024-01-25:
> You mentioned constraints, are these constraints like velocity or specific to accuracy? Will decreasing the resolution cause motions to be less accurate?

Don't worry too much about accuracy here.

> Also, if I switch to rr1.0, my arm subsystem uses RR0.5.6's MotionProfile class. Would I need to copy in the Kotlin function for that?

You can use https://rr.brott.dev/docs/v1-0-0-beta6/core/kdoc/core/com.acmerobotics.roadrunner/constant-profile.html or copy the old code. Whatever floats your boat.

> Also -- I'm looking at this Action system. If I need to run a PID loop or something similar, should I still create an Action and just run that Action within the loop?

You can write your own action or use whatever FSM/asynchronous system you have already.

SharkDjokovic, 2024-01-26:
Alright sounds good. I'm looking at the FTCLib sample command written in https://rr.brott.dev/docs/v1-0/guides/ftclib-commands/
Assuming I use this for my drive command like so:
```java
                Action movement1Left = drive.actionBuilder(drive.pose)
                        .splineToConstantHeading(new Vector2d(26.03, 42.57), Math.toRadians(270.00))
                        .splineToConstantHeading(new Vector2d(26.03, 53.19), Math.toRadians(270.00))
                        .splineTo(new Vector2d(48.67, 36.65), Math.toRadians(0.00))
                        .build();

                CommandScheduler.getInstance().schedule(
                        new SequentialCommandGroup(
                                new ActionCommand(movement1Left) //error on this line since I'm not specifying any requirements
                        )
                );
```
Requirements can't be left empty but I don't believe I have any. What is generally used in the requirements section?

rbrott, 2024-01-26:
> Each command should declare any subsystems it controls as requirements. This backs the scheduler’s resource management mechanism, ensuring that no more than one command requires a given subsystem at the same time. This prevents situations such as two different pieces of code attempting to set the same motor controller to different output values.

(from https://docs.wpilib.org/en/stable/docs/software/commandbased/commands.html#getrequirements)

If you don't care about the guarantees provided by requirements, you can give it an empty set.

SharkDjokovic, 2024-01-27:
I'm trying to copy the Kotlin files in and I'm running into this issue:
Cannot inline bytecode built with JVM target 1.8 into bytecode that is being built with JVM target 1.6. Please specify proper '-jvm-target' option
I understand that this is a Kotlin update issue, and I searched it up on StackOverflow. None of those helped me so I was wondering how to resolve.

SharkDjokovic, 2024-02-11:
This has persisted as an issue for us for RR0.5.6. Changing the resolution to any number has not changed anything, unfortunately. We ran a Profiler to check CPU/Memory. The image is attached.
[image]
CPU spiked at 53% and 253mb Memory. Is there a fix for this?

I'm wondering if this is a FTCLib/RR issue. I tried directly creating trajectory sequences in init without using the subsystem, and it still lagged out until an eventual "stuck in stop."

rbrott, 2024-02-11:
> We ran a Profiler to check CPU/Memory. The image is attached. 

This image unfortunately is not very helpful. I need more granular information about what the CPU is executing. See [this section](https://developer.android.com/studio/profile/cpu-profiler#overview) on the timeline view. I can be most helpful if you [export a trace](https://developer.android.com/studio/profile/export-traces) and attach it here for me to take a look. Make sure you capture the trajectory generation process in the trace. I would start the trace in Android Studio, initialize the op mode with the trajectories, and let it collect data for 15s before stopping.

> I'm wondering if this is a FTCLib/RR issue.

Seems unlikely given that you can comment out a few trajectories and have everything run fine. (As an aside, one of the trajectories you commented out is likely the problem. Consider trying to reproduce with one of them in a smaller op mode with less going on.)

SharkDjokovic, 2024-02-13:
So I'm trying to export the trace now, but it's not an option. I right-clicked on the CPU part. Although it's an "option," it doesn't let me select it. I've also tried going through the sessions pane. It only gives me an option to delete the trace there.

Since the control hub is Android 7 (I think?), it doesn't allow me to record the trace...

rbrott, 2024-02-19:
Maybe you need to follow these instructions: https://developer.android.com/studio/profile#advanced-profiling

You can also (1) make a small example without dependencies on your code and have me profile (like what happened in https://github.com/acmerobotics/road-runner/issues/97#issue-2082585152) or (2) print timings in your opmode of each segment individually to find the problem segment.

closed 2025-01-12 by rbrott