acmerobotics/road-runner-quickstart #357: Angular Ramp Logger IMU Error

nateschmelkin, 2024-01-25:
### RR FTC Version

0.1.11

### Observed Behavior

I am currently in the process of tuning our robot on RR 1.0. Our control hub is mounted with the logo facing left and the USB port facing forward. The angular ramp logger throws an error saying that Y Y-axis rotation is greater than Z Z-axis. If the IMU were un-initialized, this would make sense, but I changed the orientation in the MecanumDrive class. Is it possible that the IMU is broken or maybe not actually initializing? We did purchase Nav2X in the past and could potentially use that instead but would prefer not to if using an external gyroscope causes a lot of additional implementation.
[image]
[image]

### Tuning Files

package org.firstinspires.ftc.teamcode;

import androidx.annotation.NonNull;

import com.acmerobotics.dashboard.canvas.Canvas;
import com.acmerobotics.dashboard.config.Config;
import com.acmerobotics.dashboard.telemetry.TelemetryPacket;
import com.acmerobotics.roadrunner.*;
import com.acmerobotics.roadrunner.AngularVelConstraint;
import com.acmerobotics.roadrunner.DualNum;
import com.acmerobotics.roadrunner.HolonomicController;
import com.acmerobotics.roadrunner.MecanumKinematics;
import com.acmerobotics.roadrunner.MinVelConstraint;
import com.acmerobotics.roadrunner.MotorFeedforward;
import com.acmerobotics.roadrunner.Pose2d;
import com.acmerobotics.roadrunner.Pose2dDual;
import com.acmerobotics.roadrunner.ProfileAccelConstraint;
import com.acmerobotics.roadrunner.Time;
import com.acmerobotics.roadrunner.TimeTrajectory;
import com.acmerobotics.roadrunner.TimeTurn;
import com.acmerobotics.roadrunner.TrajectoryActionBuilder;
import com.acmerobotics.roadrunner.TurnConstraints;
import com.acmerobotics.roadrunner.Twist2dDual;
import com.acmerobotics.roadrunner.Vector2d;
import com.acmerobotics.roadrunner.VelConstraint;
import com.acmerobotics.roadrunner.ftc.DownsampledWriter;
import com.acmerobotics.roadrunner.ftc.Encoder;
import com.acmerobotics.roadrunner.ftc.FlightRecorder;
import com.acmerobotics.roadrunner.ftc.LynxFirmware;
import com.acmerobotics.roadrunner.ftc.OverflowEncoder;
import com.acmerobotics.roadrunner.ftc.PositionVelocityPair;
import com.acmerobotics.roadrunner.ftc.RawEncoder;
import com.qualcomm.hardware.lynx.LynxModule;
import com.qualcomm.hardware.rev.RevHubOrientationOnRobot;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.hardware.DcMotorEx;
import com.qualcomm.robotcore.hardware.DcMotorSimple;
import com.qualcomm.robotcore.hardware.HardwareMap;
import com.qualcomm.robotcore.hardware.IMU;
import com.qualcomm.robotcore.hardware.VoltageSensor;

import org.firstinspires.ftc.robotcore.external.navigation.AngleUnit;
import org.firstinspires.ftc.teamcode.messages.DriveCommandMessage;
import org.firstinspires.ftc.teamcode.messages.MecanumCommandMessage;
import org.firstinspires.ftc.teamcode.messages.MecanumEncodersMessage;
import org.firstinspires.ftc.teamcode.messages.PoseMessage;
import org.firstinspires.ftc.teamcode.subsystems.settings.ConfigInfo;

import java.lang.Math;
import java.util.Arrays;
import java.util.LinkedList;
import java.util.List;

@Config
public final class MecanumDrive {
    public static class Params {
        // IMU orientation
        // TODO: fill in these values based on
[... 401 lines cut ...]
    }

    private static void drawRobot(Canvas c, Pose2d t) {
        final double ROBOT_RADIUS = 9;

        c.setStrokeWidth(1);
        c.strokeCircle(t.position.x, t.position.y, ROBOT_RADIUS);

        Vector2d halfv = t.heading.vec().times(0.5 * ROBOT_RADIUS);
        Vector2d p1 = t.position.plus(halfv);
        Vector2d p2 = p1.plus(halfv);
        c.strokeLine(p1.x, p1.y, p2.x, p2.y);
    }

    public TrajectoryActionBuilder actionBuilder(Pose2d beginPose) {
        return new TrajectoryActionBuilder(
                TurnAction::new,
                FollowTrajectoryAction::new,
                beginPose, 1e-6, 0.0,
                defaultTurnConstraints,
                defaultVelConstraint, defaultAccelConstraint,
                0.25, 0.1
        );
    }
}

rbrott, 2024-01-25:
The IMU orientation seems correct at first glance. You should try running `ConceptExploringIMUOrientation` and find an orientation that makes yaw increase (and yaw velocity > 0) when rotating the robot counter clockwise on the ground. You can then use that orientation for tuning, and you should report an issue against the FTC SDK if that orientation doesn't seem consistent with the IRL placement of your hub.

nateschmelkin, 2024-01-25:
Thank you for this advice. I will test it and let you know if it works.
Hopefully, it does!

closed 2024-09-09 by nateschmelkin

TrentLeitzell, 2025-01-10:
Hey, currently have a similar issue. How did you fix it?

nateschmelkin, 2025-01-10:
Turned out to be an error in the control hub's local timing. It pulls from
the latest UNIX timecode log and since the control hub had a hardware error
where it jumped its time, I kept pulling from the wrong time. I suggest you
connect your CH to your computer and then look for the logs folder of
something similar and try to reset those. Alternative would be to factory
reset the hub I believe as the log will reset. Let me know if this helps
and I can give a more detailed response if not.

TrentLeitzell, 2025-01-10:
Thank you! I'll try that and let you know if it works or not.

jpdagostino, 2025-01-12:
What is the location of the UNIX timecode log? I can't seem to figure this out. I also tried using the IMU on the expansion hub but still get a similar error:
"More rotation about the +x-axis than the +z-axis. Fix the IMU orientation and run again."

Nothing I have tried seems to resolve this, but it's probably really obvious.

nateschmelkin, 2025-01-13:
Your best bet will be to factory reset the CH and then redo the tests. This
should fix the issue since the logs are stored locally on the hub I believe.

seweed3, 2025-01-14:
Yes we had this issue too. We reset the CH but our imu was being taken from our expansion hub. Everything was fine, IMU orientation correct yet it said more rotation in y than z which didn't make sense because the z was going up by a lot when rotating manually. Not sure what to do....

rbrott, 2025-01-14:
I made an issue here for the time stuff: https://github.com/acmerobotics/road-runner-ftc/issues/12

@seweed3 You should open a new issue for your problem and include the tuning file.

seweed3, 2025-01-14:
Yes, I will I'm just waiting for robot logs before i can submit it.