acmerobotics/road-runner-quickstart #359: Cannot strafe at a diagonal or to a heading

nash-pillai, 2024-01-27:
### RR FTC Version

0.1.11

### Observed Behavior

[image]
[image]
Strafing while turning and strafing at a diagonal goes in the wrong direction, but the robot knows where it is

The code we are running:
```kotlin
drive.actionBuilder(beginPose)
.strafeToSplineHeading(Vector2d(12.0, 38.0), -TAU / 4)
.build()
```
```kotlin
drive.actionBuilder(beginPose)
.strafeTo(Vector2d(17.0, 38.0))
.build()
```

### Tuning Files

<details>
<summary>MecanumDrive.java:</summary>

```java
package org.firstinspires.ftc.teamcode;

import androidx.annotation.NonNull;

import com.acmerobotics.dashboard.canvas.Canvas;
import com.acmerobotics.dashboard.config.Config;
import com.acmerobotics.dashboard.telemetry.TelemetryPacket;
import com.acmerobotics.roadrunner.AccelConstraint;
import com.acmerobotics.roadrunner.Action;
import com.acmerobotics.roadrunner.Actions;
import com.acmerobotics.roadrunner.AngularVelConstraint;
import com.acmerobotics.roadrunner.DualNum;
import com.acmerobotics.roadrunner.HolonomicController;
import com.acmerobotics.roadrunner.MecanumKinematics;
import com.acmerobotics.roadrunner.MinVelConstraint;
import com.acmerobotics.roadrunner.MotorFeedforward;
import com.acmerobotics.roadrunner.Pose2d;
import com.acmerobotics.roadrunner.Pose2dDual;
import com.acmerobotics.roadrunner.PoseVelocity2d;
import com.acmerobotics.roadrunner.PoseVelocity2dDual;
import com.acmerobotics.roadrunner.ProfileAccelConstraint;
import com.acmerobotics.roadrunner.Rotation2d;
import com.acmerobotics.roadrunner.Time;
import com.acmerobotics.roadrunner.TimeTrajectory;
import com.acmerobotics.roadrunner.TimeTurn;
import com.acmerobotics.roadrunner.TrajectoryActionBuilder;
import com.acmerobotics.roadrunner.TurnConstraints;
import com.acmerobotics.roadrunner.Twist2dDual;
import com.acmerobotics.roadrunner.Vector2d;
import com.acmerobotics.roadrunner.VelConstraint;
import com.acmerobotics.roadrunner.ftc.DownsampledWriter;
import com.acmerobotics.roadrunner.ftc.Encoder;
import com.acmerobotics.roadrunner.ftc.FlightRecorder;
import com.acmerobotics.roadrunner.ftc.LynxFirmware;
import com.acmerobotics.roadrunner.ftc.OverflowEncoder;
import com.acmerobotics.roadrunner.ftc.PositionVelocityPair;
import com.acmerobotics.roadrunner.ftc.RawEncoder;
import com.qualcomm.hardware.lynx.LynxModule;
import com.qualcomm.hardware.rev.RevHubOrientationOnRobot;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.hardware.DcMotorEx;
import com.qualcomm.robotcore.hardware.DcMotorSimple;
[... 515 lines cut ...]
                new Vector2dDual<>(
                        new DualNum<Time>(new double[]{
                                (PARAMS.par0YTicks * par1PosDelta - PARAMS.par1YTicks * par0PosDelta) / (PARAMS.par0YTicks - PARAMS.par1YTicks),
                                (PARAMS.par0YTicks * par1PosVel.velocity - PARAMS.par1YTicks * par0PosVel.velocity) / (PARAMS.par0YTicks - PARAMS.par1YTicks),
                        }).times(inPerTick),
                        new DualNum<Time>(new double[]{
                                (PARAMS.perpXTicks / (PARAMS.par0YTicks - PARAMS.par1YTicks) * (par1PosDelta - par0PosDelta) + perpPosDelta),
                                (PARAMS.perpXTicks / (PARAMS.par0YTicks - PARAMS.par1YTicks) * (par1PosVel.velocity - par0PosVel.velocity) + perpPosVel.velocity),
                        }).times(inPerTick)
                ),
                new DualNum<>(new double[]{
                        (par0PosDelta - par1PosDelta) / (PARAMS.par0YTicks - PARAMS.par1YTicks),
                        (par0PosVel.velocity - par1PosVel.velocity) / (PARAMS.par0YTicks - PARAMS.par1YTicks),
                })
        );

        lastPar0Pos = par0PosVel.position;
        lastPar1Pos = par1PosVel.position;
        lastPerpPos = perpPosVel.position;

        return twist;
    }
}
```
</details>

rbrott, 2024-01-28:
Does the same thing happen if the velocity is much lower? Can you link a robot log?

nash-pillai, 2024-01-28:
It continued to happen even when we lowered to:
```java
 public double maxWheelVel = 20;
 public double minProfileAccel = -10;
 public double maxProfileAccel = 10;
```
I can send robot logs tomorrow

nash-pillai, 2024-01-28:
[robotControllerLog.txt](https://github.com/acmerobotics/road-runner-quickstart/files/14077997/robotControllerLog.txt)
[2024_01_28__17_30_28_038__MenuAuto.log](https://github.com/acmerobotics/road-runner-quickstart/files/14077998/2024_01_28__17_30_28_038__MenuAuto.log)

I think these are the files you are looking for. Let me know if you want something else

rbrott, 2024-01-29:
Yup, the second one is the log I need, thanks. Here's a summary of its contents
```
THREE_DEAD_WHEEL_PARAMS
  par0YTicks: -962.0
  par1YTicks: 912.0
  perpXTicks: -2133.0

MECANUM_PARAMS
  axialGain: 0.0
  axialVelGain: 0.0
  headingGain: 4.32
  headingVelGain: 0.59
  inPerTick: 0.0030135610246107484
  kA: 1e-06
  kS: 0.9589658474985132
  kV: 0.0005588422234744558
  lateralGain: 0.0
  lateralInPerTick: 0.00018869762188191035
  lateralVelGain: 0.0
  logoFacingDirection: UP
  maxAngAccel: 3.141592653589793
  maxAngVel: 3.141592653589793
  maxProfileAccel: 20.0
  maxWheelVel: 40.0
  minProfileAccel: -20.0
  trackWidthTicks: 4432.299
  usbFacingDirection: RIGHT

Channel                Messages         Hz
TARGET_POSE                  61      20.01
ESTIMATED_POSE               62      20.23
DRIVE_COMMAND                62      20.26
MECANUM_COMMAND              62      20.92
```
Right off the bat, `lateralInPerTick` seems very suspicious. It should be at most like 25% less than `inPerTick` and not an order of magnitude off. Can you try tuning that again and sharing the results here (including the JSON)?

nash-pillai, 2024-02-02:
Thank you so much! I'm not how we messed it up in the first place, but re-tuning lateralInPerTick fixed our problems

closed 2024-02-02 by nash-pillai