acmerobotics/road-runner-quickstart #364: Meepmeep values suck (the tuning is alright - must be)

tymDahmani, 2024-02-16:
### RR FTC Version

0.1.11

### Observed Behavior

our tuning is fine, we tuned and it the robot worked well, but now at the last week before our qualifiers (our qualifiers day 1 is literally today in few hours) the robot's movements arent correct, we used meepmeep values but they didnt work at all! they are very different from reality and we had to tune the movements of some movements manually! we have no time left to tune all of it manually so i hope anybody can help please!!!

### Tuning Files

our full team code, u can view all our tuning code here: https://github.com/Abdulmalek-HoM/QuantaTeam-Autonomous-Codes/tree/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode

rbrott, 2024-02-16:
Did you change something about your robot? Sounds like your tuning is bad after all.

tymDahmani, 2024-02-16:
really.....? ahaaa ima cry fr, its been working fine!!!!