acmerobotics/road-runner-quickstart #365: "Cannot resolve symbol" in AcmeRobotics

RyanGOttignon, 2024-02-18:
### RR FTC Version

0.1.12

### Observed Behavior

Upon downloading the newest version of the quickstart and opening it in Android Studio, all `com.acmerobotics.roadrunner` classes fail to resolve.

[image]

This is identical to #201, but no answer was given there.

rbrott, 2024-02-18:
Have you tried building the project or doing a Gradle sync?

RyanGOttignon, 2024-02-19:
I have tried doing multiple Gradle syncs, and all had this same result. I will try building the project tomorrow, though I suspect it'll just throw the same exception as Android Studio hints at.

RyanGOttignon, 2024-02-20:
Building works! I'm not sure why the errors appear in Android Studio, but it works while building.

Thank you!

closed 2024-02-20 by RyanGOttignon

niceeffort, 2024-02-28:
We had this problem yesterday. The issue was that we could not download the POM files. We solved it by installing [CloudFlare Warp](https://cloudflarewarp.com/) as suggested in this thread: https://www.reddit.com/r/FTC/comments/1at3sb0/is_the_acmerobotics_repo_down_i_cant_download/