acmerobotics/road-runner-quickstart #371: Import Statements in MecanumDrive.java and TankDrive.java having numerous errors

TheAwesomeKid02, 2024-03-03:
### RR FTC Version

0.1.13

### Observed Behavior

When I open `MecanumDrive.java` and `TankDrive.java`, I can see numerous importing errors. Upon further inspection, I found out that the files don't even exist! The import is like `import com.acmerobotics.roadrunner.Action` (This is an example) but the file does not exist. Same with numerous other classes like `AccelConstraint`, `Arclength`, `VelConstraint`, etc.

rbrott, 2024-03-04:
Sounds similar to #365. Can you try building the project with the hammer and see if any errors pop up there (click the hammer in the top bar or run `./gradlew build -x lint`)?

TheAwesomeKid02, 2024-03-04:
Building finished with zero errors. I am connected to regular wifi yet the problem still persists.

TheAwesomeKid02, 2024-03-04:
[image]
This is the error of the imports.

TheAwesomeKid02, 2024-03-04:
Could someone please help me?

ftcaarya, 2024-03-05:
> Could someone please help me? Or you can recommend someone who can help me.

You could try to clone a new quickstart and transfer all the edits in there this might help

TheAwesomeKid02, 2024-03-06:
> > Could someone please help me? Or you can recommend someone who can help me.
> 
> You could try to clone a new quickstart and transfer all the edits in there, we had a problem like this and it was resolved by cloning a new quickstart repo.

What edits would I transfer and what new quickstart? I am just asking for further clarification.

ftcaarya, 2024-03-06:
> What edits would I transfer and what new quickstart? I am just asking for further clarification.

All the tuning values that changed and all the opmodes that you created

TheAwesomeKid02, 2024-03-06:
I have not made any edits on the repo yet. I just opened the files to check them out and I found the errors.

ftcaarya, 2024-03-06:
> I have not made any edits on the repo yet. I just opened the files to check them out and I found the errors.

Then just clone a new quickstart repo

TheAwesomeKid02, 2024-03-10:
The problem persists. I tried re-cloning the repo and re-building.

vidhuthalla, 2024-04-17:
Have you found a solution I am having the same problem

vidhuthalla, 2024-04-17:
> The problem persists. I tried re-cloning the repo and re-building.

Have you found a solution I am having the same problem

TheAwesomeKid02, 2024-04-17:
> > The problem persists. I tried re-cloning the repo and re-building.
> 
> 
> 
> 
> 
> Have you found a solution I am having the same problem
> 
> 
> 
> 

Unfortunately, I have not found a solution for this.

TullyNYGuy, 2024-06-10:
Same issue here. I have a fresh, unmodified download of the quickstart. Invalidating the cache, gradle sync and build (i.e hammer) did not resolve it. The code has "cannot resolve symbol" for all of the com.acmerobotics.roadrunner classes but I have no build errors. That seems odd to me.

TullyNYGuy, 2024-06-10:
After much messing around, I decided to upgrade gradle from 7.2.0 to 7.3.1. That did not help. Then I upgraded Android Studio from Dolphin to Jellyfish. That fixed the issue.

TheAwesomeKid02, 2024-07-19:
> After much messing around, I decided to upgrade gradle from 7.2.0 to 7.3.1. That did not help. Then I upgraded Android Studio from Dolphin to Jellyfish. That fixed the issue.

Alright, I will try this. I will let you know if this works.

TheAwesomeKid02, 2024-07-21:
Good news...
It appears that this error is not present for Android Studio versions after Jellyfish?? (correct me if I am wrong). I upgraded my Android Studio from Dolphin to Koala and it works fine now.

closed 2024-07-21 by TheAwesomeKid02