acmerobotics/road-runner-quickstart #372: Failed Requirement during ManualFeedbackTuner or any Opmode created

ftcaarya, 2024-03-05:
### RR FTC Version

0.1.13

### Observed Behavior

After all tuning has been done for drive encoders, I was running the ManualFeedbackTuner opmode that was given with no edits to the file. When I ran it though I get a error saying "Failed Requirement" And the team code line says a line, I delete that line and run again it shows a different line with the same error. All these lines are in a trajectory builder btw.

ftcaarya, 2024-03-07:
@rbrott

rbrott, 2024-03-07:
Can you please post the full stacktrace?

ftcaarya, 2024-03-07:
[image]
[image]

rbrott, 2024-03-08:
Can you attach a robot log?

ftcaarya, 2024-03-08:
> Can you attach a robot log?

What does that mean? Like the telemetry or the robots behavior or logcat?

rbrott, 2024-03-09:
Go to `http://192.168.43.1:8080/logs` and download one of the files.

ftcaarya, 2024-04-12:
[2024_03_01__19_32_30_036__ManualFeedbackTuner.log](https://github.com/acmerobotics/road-runner-quickstart/files/14964490/2024_03_01__19_32_30_036__ManualFeedbackTuner.log)

ftcaarya, 2024-04-12:
> Go to `http://192.168.43.1:8080/logs` and download one of the files.

@rbrott I linked it

rbrott, 2024-04-13:
So your params are
```
MECANUM_PARAMS
  axialGain: 0.0
  axialVelGain: 0.0
  headingGain: 0.0
  headingVelGain: 0.0
  inPerTick: 0.02258823529411765
  kA: 0.0005
  kS: 1.466179965890956
  kV: 0.004050112092712909
  lateralGain: 0.0
  lateralInPerTick: 0.0
  lateralVelGain: 0.0
  logoFacingDirection: BACKWARD
  maxAngAccel: 3.141592653589793
  maxAngVel: 3.141592653589793
  maxProfileAccel: 50.0
  maxWheelVel: 50.0
  minProfileAccel: -30.0
  trackWidthTicks: 1330.9955862311713
  usbFacingDirection: UP
```
`lateralInPerTick` should not be 0.0. Follow the tuning instructions to calculate its value.

ftcaarya, 2024-04-13:
> instructions

In my MecanumDrive.java class I have lateralInPerTick set as 24/1025. Am I missing something?

ftcaarya, 2024-04-13:
> So your params are
>
> ```
> MECANUM_PARAMS
>   axialGain: 0.0
>   axialVelGain: 0.0
>   headingGain: 0.0
>   headingVelGain: 0.0
>   inPerTick: 0.02258823529411765
>   kA: 0.0005
>   kS: 1.466179965890956
>   kV: [...]

I had the letralInPerTick set but I was doing integer division instead of double division it was set as 24/1025 where it should be 24.0/1025 sorry for the trouble and thanks

EthanHu29, 2024-05-25:
@rbrott , Any tool we can make the log files readable for human being? Or we have to write one ourselvles? Any suggestion? Thanks!

closed 2025-01-12 by rbrott