acmerobotics/road-runner-quickstart #373: Path tangent orthogonal to the y-axis

RyanGOttignon, 2024-03-09:
### RR FTC Version

0.1.13

### Observed Behavior

When using `lineToY` statements split by a `turn`, an exception is thrown: `Path tangent orthogonal to the y-axis, try using lineToX() instead`

The expected behavior is a forward movement, a turn, and then another forward movement.

Full code:

```
// QTR_FIELD and STG_Y_IN are constants that define the starting position of the robot; they can be any value
MecanumDrive drive = new MecanumDrive(hardwareMap, new Pose2d(QTR_FLD, STG_Y_IN, Math.toRadians(90.0)));

Action forwardMove;
forwardMove = drive.actionBuilder(drive.pose)
                           .lineToY(STG_Y_IN + 30)
                           .turn(Math.toRadians(90))
                           .lineToY(STG_Y_IN + 49.5)
                           .build();
```

#325 advises to use `setTangent`, but results are odd. An argument of `0.0` obviously continues to throw the error; `90.0` causes the second movement to go 45 degrees to the left of the expected movement forward; `180.0` causes the second movement to go 45 degrees to the right of the expected movement forward; `135.0` (trying to average to two) goes straight left of the expected movement forward.

I have a suspicion that `setTangent` is the solution here, but I am struggling to find a value that simply goes straight. Perhaps elaborating on what it does would be handy.

rbrott, 2024-03-09:
`turn()` moves the tangent of the path with it, so it sounds like you want to set the tangent to 90deg again right after it completes: https://rr.brott.dev/playground/?b8c6d36f4f8b9b14. Not sure why your earlier attempt with 90deg went wrong.

RyanGOttignon, 2024-03-11:
Welp, there's a lesson. (Thank you!)

Note to future programmers who stumble across this: `turn()` takes an angle in radians, per the docs. Make sure you're in radians. :p

closed 2024-03-11 by RyanGOttignon