acmerobotics/road-runner-quickstart #384: Adding more TrajectoryBuilders in the Builder Reference

ArushYadlapati, 2024-06-29:
### RR FTC Version

1.0

### Observed Behavior

The current [Builder Reference](https://rr.brott.dev/docs/v1-0/builder-ref/) in the official Road Runner 1.0 docs only has a few TrajectoryBuilder methods, and does not explain them very well in depth. I am currently working on a complete reference for more methods in the TrajectoryBuilder class for Road Runner 1.0.

However, in order to create this, I need to know how rr.brott.dev was created, and if possible, make it open source to make contributing to it easier, similar to LearnRoadRunner. If I could get a copy of the code for the current Builder Reference page that I can add onto, that would also work as well. Finally, I am trying to embed LaTeX into this, so I would appreciate any pointers on doing so (sending me the current Builder Reference file should also work).

Below is this list of TrajectoryBuilder's that I will be adding references to (using the [Road Runner 1.0 Playground](https://rr.brott.dev/playground/)).
```             
                 .waitSeconds(time)
                 .turn(angle)
                 .turnTo(heading)
                 .setTangent(r)
                 .setReversed(reversed)

                 .lineToY(y)/.lineToYConstantHeading(y)
                 .lineToYLinearHeading(y, heading)
                 .lineToYSplineHeading(y, heading)

                 .lineToX(x)/.lineToYConstantHeading(x)
                 .lineToXLinearHeading(x, heading)
                 .lineToXSplineHeading(x, heading)

                 .strafeTo(new Vector2d(x, y))/.strafeToConstantHeading(new Vector2d(x, y))
                 .strafeToLinearHeading(new Vector2d(x, y), Math.toRadians(heading))
                 .strafeToSplineHeading(new Vector2d(x, y), Math.toRadians(heading))

                 .splineTo(new Vector2d(x, y), tangent)
                 .splineToConstantHeading(new Vector2d(x, y), tangent)
                 .splineToSplineHeading(new Pose2d(x, y, Math.toRadians(heading)), tangent)
                 .splineToLinearHeading(new Pose2d(x, y, Math.toRadians(heading)), tangent)
```

rbrott, 2024-07-08:
The main source file for this page is [index.md](https://github.com/user-attachments/files/16124781/index.md).

In general I want to have more editorial control over the docs, and I don't want to accept PRs. I may make the website source available in some form but it's not fully open source. I know not everyone is happy with this form of unattributed, ad hoc contribution, so I want you to be aware before you put any more time into this.

ArushYadlapati, 2024-07-09:
I thought this might be the case, so I also made it available as a new recipe in the [Cookbook](https://arushyadlapati.github.io/cookbook/roadrunner_10/complete_trajectorybuilder_reference.html).

closed 2025-01-12 by rbrott