acmerobotics/road-runner-quickstart #388: Parallel Wheel 0 Regression Producing Unusual Results, Unsure of which points to exclude.

AJain862, 2024-07-18:
### RR FTC Version

0.1.12

### Observed Behavior

When I run the RR Dead Wheel Angular Ramp Regression, while using a 2 dead wheel setup (picture attached) my parallel wheel 0 regression looks very unusual I have attached the tuning file below. I am unsure of which points to exclude. I have tried, but I am not sure if this is normal, or if I have even done it correctly. Any help would be greatly appreciated. Thanks!

### Tuning Files

[image]

### Robot Logs

Not sure which logs are relevant, if you could let me know I will attach them in a response.

rbrott, 2024-07-22:
Excluding points won't help here. Something is wrong with the hardware setup that's causing the wheels to bounce or something. I would try springing the wheel more to the ground and making sure the whole apparatus is as rigid as possible.

AJain862, 2024-07-22:
Ok, I will check the hardware when I get a chance, however to note, I have been running this hardware set up for the entire season with roadrunner 0.5, so i highly doubt that the hardware is an issue or something would have been seen throughout the season when using rr 0.5. I also continued using the current points i excluded and nothing is extremely wrong, however the lateral accuracy is not that great (not sure if that is related to this tho)

closed 2025-01-12 by rbrott