acmerobotics/road-runner-quickstart #391: `SyntaxError: Unexpected end of JSON input` in `ForwardRampLogger`

Marlstar, 2024-08-07:
### RR FTC Version

1.0.0

### Observed Behavior

When attempting to open the latest logged data for [`ForwardRampLogger`](http://192.168.43.1:8080/tuning/forward-ramp.html), an error occurs due to the raw json file being randomly cut off.
[image]
When opening the JSON file, this is evident, as it appears that the file was still being written to before abruptly stopping without completing the JSON formatting.

[image]

### Tuning Files

File: [forward-ramp-1722318233797 (1).json](https://github.com/user-attachments/files/16505245/forward-ramp-1722318233797.1.json)

### Robot Logs

[2024_08_02__13_19_16_399__ManualFeedbackTuner.log](https://github.com/user-attachments/files/16517605/2024_08_02__13_19_16_399__ManualFeedbackTuner.log)

rbrott, 2024-08-09:
As you noticed the file is incomplete. I would just try again, maybe stopping it sooner.

Marlstar, 2024-08-09:
Hello,
I did actually try running it a few times, including shorter runs, but got the same results. This is strange as I have gone through this tuning stage multiple times before without issue. I will try a few more times just to be sure, but I do not believe it is an issue on my end.

Thanks!

Marlstar, 2024-09-02:
Hello! Just an update, I am moving back to RR 0.5 for the time being given that it is much more well-known and the community knowledge is much greater. I'll close the issue for now. Thanks :)

closed 2024-09-02 by Marlstar