acmerobotics/road-runner-quickstart #409: How can I merge the latest FTC Robot Controller with our own team's fork?

Quack4321, 2024-09-25:
We need the latest version of FTC Robot Controller on our own fork of the repository. Is there a way to do this without replacing the entire project? Not sure where else to ask this question, I'm not very familiar with all this.

rbrott, 2024-09-25:
https://rr.brott.dev/docs/v1-0/installation/

Quack4321, 2024-09-25:
Sorry if the qustion was unclear, I am asking if there is a way to put the new FTC Robot Controller files into an existing fork of road runner quickstart without creating another project, not how to bring an existing non road runner project into the road runner quickstart. But actually it's probably easy enough to just create another project and put teamcode in there, so that's what I'll do, thanks

rbrott, 2024-09-25:
You can also try these steps: https://ftc-docs.firstinspires.org/en/latest/programming_resources/tutorial_specific/android_studio/fork_and_clone_github_repository/Fork-and-Clone-From-GitHub.html?highlight=pull#updating-your-fork-and-local-clone

ethansocal, 2024-09-25:
For our repository, we were able to press this "Sync fork" button, and click a button that said something like "Update project". If you have edited the files that were updated upstream, you might need to create a pull request, and merge it manually.
[image]

Quack4321, 2024-09-25:
Hey thanks I literally JUST figured this out and synced my project a minute ago. All the files that we contributed were separate from the updated roadrunner project. I'm still figuring out GitHub, but pressing sync fork solved my problem I think. Feeling stupid with how simple that was...

closed 2024-09-25 by Quack4321