acmerobotics/road-runner-quickstart #410: Road Runner tuning questions

rbrott, 2024-09-25:
Team 25780 emailed with some questions. I'm going to answer them here with the hope that more people will stumble across this information.

> We recently started using the RoadRunnerDocs to use RoadRunner. We installed the road-runner-quickstart file through GitHub, which we found on the [rr.brott.dev](http://rr.brott.dev/) website. It was this GitHub link: https://github.com/acmerobotics/road-runner-quickstart. We followed the installation steps and started our tuning process. I started the tuning process but the first problem we encountered was that we couldn't see a line of code. For background knowledge, we are using a mecanum drive wheelbase and 3 dead wheels. We followed this step: 
> Three (dead) wheel: Change the right-hand-side of localizer = to new ThreeDeadWheelLocalizer(hardwareMap, PARAMS.inPerTick). The code expects the two parallel encoders to be named "par0" and "par1" and the perpendicular one to be named "perp".
> However, we couldn't find the "localizer = " line. Is this located in the TuningOpModes file, or is it located somewhere else?

It's located [here](https://github.com/acmerobotics/road-runner-quickstart/blob/5f35f4c22c1ae7c0be5b35da0961c8f3a181ad31/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/MecanumDrive.java#L240) for mecanum and [here](https://github.com/acmerobotics/road-runner-quickstart/blob/5f35f4c22c1ae7c0be5b35da0961c8f3a181ad31/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/TankDrive.java#L247) for tank.

> We decided to move on and we encountered another problem during this step: 

Do not skip steps! You are much better off asking in the FTC Discord or even opening a new issue here.

> Those with mecanum drives should use MecanumDirectionDebugger to make sure all the directions are correct. The op mode uses the following button mappings:
> 
> We couldn't find the MecanumDirectionDebugger file. Did we skip a step to download additional files or did we download from the wrong GitHub repository? After that, we couldn't find the file named DeadWheelDirectionDebugger.
> 
> We then moved on to the FTC Dashboard part. However, we couldn't find the ForwardPushTest, the LateralPushTest, or the ForwardRampLogger. Is there another repository that we need to download files from, or are they hidden somewhere in the files that we already downloaded?

These op modes are not included directly in the quickstart, but they should appear on the Driver Station if you've setup everything correctly.

samhithinavolu, 2024-09-28:
Hi, I am part of team 25780, CyberChaos, and I have some more questions on the tuning process:

I have doubts about the values I got for the inPerTick, lateralInPerTick, trackWidthTicks, kV, and kS. I did the ForwardPushTest and got 0.0059125 for the inPerTick. The distance traveled was 120 inches and the ticks I got from the encoders was 20295.9830867 ticks. 

Next, I did the ForwardRampLogger and got kS = 1.7811092303367975 and kV = -0.0003004190059495368.
[image]
I'm not sure if these values are in the correct range or if the kV is too close to zero.

Then, I did the LateralRampLogger and got a lateralInPerTick of -0.0028922498586113937. I'm not sure if this is in the correct range or if it needs to be positive. 
[image]

Lastly, I did the AngularRampLogger and got -7815.355690456733 for trackWidthTicks. I'm not sure if this has to be a positie number. Here are the images of the graphs:
[image]
[image]
[image]
[image]
 
Thank You, Team 25780, CyberChaos.

rbrott, 2024-09-28:
> I have doubts about the values I got for the inPerTick, lateralInPerTick, trackWidthTicks, kV, and kS. I did the ForwardPushTest and got 0.0059125 for the inPerTick. The distance traveled was 120 inches and the ticks I got from the encoders was 20295.9830867 ticks.

That seems reasonable to me. FTC motors with typical gearing and typical wheel sizes give roughly hundreds of ticks per inch (so ~0.01 in / tick).

> I'm not sure if these values are in the correct range or if the kV is too close to zero.

kV should always be positive so you have some motors to reverse. It will be close to zero though.

> Then, I did the LateralRampLogger and got a lateralInPerTick of -0.0028922498586113937. I'm not sure if this is in the correct range or if it needs to be positive.

That one should also be positive.

> Lastly, I did the AngularRampLogger and got -7815.355690456733 for trackWidthTicks. I'm not sure if this has to be a positie number.

Positive as well

And looks like one of your dead wheels is unplugged

wondermanA380, 2024-10-06:
As a follow up question, we got through ManualFeedbackTuner and are assuming that we have to change parameters values at the top of the ThreeDeadWheelLocalizer.java file but we are not completely sure. We were also wondering how we should plot the graph (ex. should we plot y and y error when we change par0yticks?) and how to validate the parameters.

closed 2025-01-12 by rbrott