acmerobotics/road-runner-quickstart #413: OctoQuad Integration

LarryMoe11, 2024-10-04:
FTC Added a new device to the official hardware library, the OctoQuad, which our team is currently using in i2c mode. When attempting to add compatability between the octoquad and roadrunner, we had created a custom class to act as a compatability layer which worked quite well in the drive programming, but we found in tuning that we had no way to access the motor controller.

My question is, if teams have no way to access the motor controller (due to a device not having one), is there any current way to integrate it with roadrunner, and if not currently available, can that be supported moving forward?

raypratt, 2024-10-08:
We have struggled with this exact issue. Some direction would be awesome!

rbrott, 2024-10-09:
As you've seen, the tuning code only supports quadratic encoders read through a rev hub. 

Is there a reason that you're using OctoQuad for your drive encoders? I understand the value when only one hub is available but otherwise I don't see the advantage.

LarryMoe11, 2024-10-10:
We are using the octoquad for encoders so we could attach the motor encoders for PID tuning and speed accuracy on all of our drive motors as well as our mechanisms.