acmerobotics/road-runner-quickstart #425: I'm getting an error for my AutoMode that nobody can figure out. I have done a lot so please help "One or more of the attached Lynx Modules has outdated firmware (It is all up to date and on the suggested 1.8.2 version)." "Expansion Hub 2: (unparseable) null" Control Hub: (unparseable null)

Leoshrotri, 2024-10-25:
### RR FTC Version

0.1.14

### Observed Behavior

I have no idea why this is popping up because I already have my control hub, expansion hub, and driver hub all up to date on the 1.8.2 version that it is suggesting. I tried REV Hardware Client or whatever its called and everything is up to date. I have no idea why no code will even run and why the error only shows up after I click stop. I have also tried to re-install the firmware on the hubs and check the wiring between everything and they are all fine. The issue still persists. Please help me figure this out. FTC discord says this is a first of a kind issue nobody can seem to figure out. I don't know why it says null if everything is there please help, this is my last option I have literally done everything.
[image]
[image]
[image]
[image]
[image]

rbrott, 2024-10-25:
You can remove this line to get it to stop crashing: https://github.com/acmerobotics/road-runner-quickstart/blob/8e7870ff1e9dfe3547b2aae60dfded00e736bc69/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/MecanumDrive.java#L212.

Not sure why the version is garbage though. You may have a more serious issue with the connection to the hubs or something.

Leoshrotri, 2024-10-25:
Is there anything wrong in my auto mode coding?

Leoshrotri, 2024-10-25:
Because the error only pops up after I press stop. nothing happens when i click start, there are no actions taking place from the coding and there is no error that shows up until I press stop. I don't know why this is or if this is irregular behavior, is the code wrong?

closed 2024-10-26 by Leoshrotri