acmerobotics/road-runner-quickstart #446: Can't find tuning files for roadrunner quickstart main branch

Triticism, 2024-11-25:
I installed Roadrunner on Android Studio and followed all the installation steps. However, when I got to tuning, I couldn't find any of the files specified on [https://rr.brott.dev/docs/v1-0/tuning]. Is there something that I missed?

rbrott, 2024-11-25:
Not all of the files are stored in the quickstart repo, though the op modes should appear on the driver station.

Patchedbug, 2024-12-12:
Where could I find the OP modes for tuning? I've modified Roadrunner to work with the new gobuilda pinpoint which replaces the internal IMU. I've fixed all of the other issues except I get an error from inside of the AngularRampLogger op mode and I need to actually see the source of this op mode in order to find the problem.

rbrott, 2024-12-12:
The rest of the tuning op modes are here: https://github.com/acmerobotics/road-runner-ftc/blob/master/RoadRunner/src/main/java/com/acmerobotics/roadrunner/ftc/Tuning.kt

closed 2025-01-12 by rbrott