acmerobotics/road-runner-quickstart #467: ManualFeedbackTuner Help

KavinSandking, 2024-12-29:
### RR FTC Version

0.1.14

### Observed Behavior

when doing the manualFeedbackTuner, we noticed that the blue line (our robot), was moving in circles on FTC dashboard, but was moving in a straight line in real life. We thought nothing of it, until, when doing the splineTest, where the robot, would not move in a proper spline, and on FTC dash, it showed the robot was moving in circles. 
[image]

rbrott, 2024-12-30:
Your localization is bad as you've diagnosed, and you should revisit that part of the tuning instructions.

KavinSandking, 2024-12-31:
We redid all the tuning instructions, and the manualFeedBackTuner was working better, the problem was that the axial, heading, and lateral gains would go into the negatives. Even if the robot was going straight in the field, the field diagram would show that robot was not moving in a straight line.

KavinSandking, 2024-12-31:
Sometimes, the robot would follow the straight path in FTC Dashboard, but after going back and forth 2 times, it would suddenly turn.

KavinSandking, 2025-01-14:
never mind, one of our dead wheels wasn't working properly.

closed 2025-01-14 by KavinSandking