acmerobotics/road-runner-quickstart #471: Tuning for RR 1.0 results in no data being shown with "N/A". Localization test also shows N/A

Leoshrotri, 2025-01-07:
### RR FTC Version

0.1.14

### Observed Behavior

I am not able to tune since it keeps on showing N/A for all the data. My code corresponds to the correct encoders and names so I can't figure out the issue. 
[image]
[image]

### Tuning Files

[forward-ramp-1813275658211 (3).json](https://github.com/user-attachments/files/18327077/forward-ramp-1813275658211.3.json)

nateschmelkin, 2025-01-18:
Having same exact issue here...

rbrott, 2025-01-23:
> The weird thing is that when I run deadwheeldirectiondebugger, it works and gives accurate data from the odometry pods. This is confusing because I think this means that the pods are giving back tracking data, which means that it shouldn't be an issue where the information is not being relayed through an encoder issue.

The tuning file indeed shows that the forward encoders are not moving. I would double check that the dead wheels are actually working.