acmerobotics/road-runner-quickstart #480: PinpointView class

bafbl, 2025-02-13:
### RR FTC Version

0.1.19

### Observed Behavior

Compiler errors on PinpointView class.

Our team is eager to use the GoBilda Pinpoint with RoadRunner and is testing the latest RR to that end. Is it possible that the PinpointView class did not get committed/pushed along with the files that reference it?

Thank you.

bafbl, 2025-02-13:
These imports are also not being found...
EncoderGroup, EncoderRef, LynxQuadratureEncoderGroup, PinpointEncoderGroup, PinpointIMU

rbrott, 2025-02-14:
Have you tried a Gradle sync? Android Studio can get confused about these kinds of things.

bafbl, 2025-02-14:
Thank you for the reminder that it's good to double-check AS. Unfortunately, it did not change anything. We still cannot find the referenced files. Just looking in github, searches for any of the following classes in roadrunner and roadrunner-quickstart find references to the classes, but not the classes themselves:  EncoderGroup, EncoderRef, LynxQuadratureEncoderGroup, PinpointEncoderGroup, PinpointIMU. Searches for PinpointView is more complicated because you're creating a dynamic subclass, but we still can't find the interface or abstract class. 

Here is a typical search result for the classes we can't find in github:

[image]

rbrott, 2025-02-15:
The files are in this repository: https://github.com/acmerobotics/road-runner-ftc

closed 2025-06-29 by rbrott