acmerobotics/road-runner-quickstart #491: Path following mode as a built-in option or even the default?

j5155, 2025-03-29:
Two of the most commonly stated cons of roadrunner are that, because of the time-based follower targeting, it can't go as fast as other followers or correct as well. It seems like the path following section of the docs has the potential to fix both of these issues. This year, my team implemented a custom follower (attached) based on the path following guide; we found it allowed our autonomous to be both faster and more reliable, and we were even able to automate our teleop deposit scoring. Even when opposing robots actively tried to get in our way, path following was able to completely deal with it. 

I am wondering if this is something that could be built into the quickstart or even replace the default follower; it seems to have huge advantages with few downsides. My implementation also definitely has a few quick fixes/hacks, it was made in-season under heavy time crunches; if you have any suggestions for improving it or properly dealing with the things I worked around I would hugely appreciate it. 

[FollowTrajectoryAsPathAction.java.txt](https://github.com/user-attachments/files/19523016/FollowTrajectoryAsPathAction.java.txt)

rbrott, 2025-03-30:
I agree that upper-level teams with robust dead wheel localization may often be better off with a path follower instead of a trajectory follower. It's great to see someone get a working path follower based in part on the implementation sketch in the docs. I would be interested in expanding that section of the docs to include a sample follower and a more complete discussion of implementation choices.

I'm not convinced this should be added to the quickstart. The existing follower and TAB infrastructure should adequately handle the needs of most teams, and the more advanced ones should be able to copy a starter path follower from the docs and iterate on it from there.

zachwaffle4, 2025-04-05:
I continue to be of the opinion that giving users options is always better; I personally used a slight modification of what Jay sent in my regional championship and it worked better than the traditional method of trajectory following. 

I believe it would be best to give teams the choice in the quickstart on how they want to follow their trajectories rather than having the other (and potentially better? as Jay said there are many advantages with no downsides that I can think of) option only be on a page of docs. I've been working on a bunch of small changes to the core library that I will PR after champs that I think will make this (and the whole TAB api) easier to work with.