acmerobotics/road-runner-quickstart #496: OtosLinearScalarTuner

hakanatas, 2025-06-12:
Hi,
I am trying to use Otos sensor. I made OTOSAngularScalarTuner and get result. But When I started OTOSLinearScalarTuner Uncorrected Distance Traveled Y and X are always zero. 

Any solution to fix it?

bubner, 2025-06-14:
> When I started OTOSLinearScalarTuner Uncorrected Distance Traveled Y and X are always zero.

This can happen if your `inPerTick` value is zero. Do confirm you've set `inPerTick` on either `MecanumDrive` or `TankDrive` to `1.0`.

LAtimes2, 2025-07-28:
All of the OTOS tuner opmodes need to call bulkRead. In road-runner-ftc Tuning.kt, each opmode needs this inside the run loop:

```
                    for (g in view.encoderGroups) {
                        g.bulkRead()
                    }
```

Another suggestion is to add gamepad control of robot to each opmode - similar to lines 607-613 of Tuning.kt.