acmerobotics/road-runner-quickstart #508: LocalizationTest Not Reporting Correct Position with Pinpoint and 2 Odopods

roperb, 2025-07-28:
### RR FTC Version

0.1.24

### Observed Behavior

Downloaded a fresh install of the 0.1.24 RR 1.0 quickstart and tuned a mechanum chassis robot going through the motor and encoder direction tests, then ForwardPushTest, ForwardRampLogger, and AngularRampLogger.  Everything worked great and numbers were reasonable for InPerTick, kV, kS, TrackWidthTicks, and the encoder positions in ticks.  All values were updated and pushed onto the robot.

At this point, the LocalizationTest should display properly, but forward and backward motion didn't register properly.  Instead of driving forward 24 inches, it said I had moved only maybe an inch or so and it wasn't consistent, so no obvious unit choice issue.

The heading was more or less fine when I turned it 360 degrees each direction.  The field view made the robot turn properly but go in a somewhat wide circle, reflecting a position change when there was none.

I downloaded the GoBildaPinPointExample and put in the proper offsets and settings and it output the expected localization for driving forward and leftward by 2 feet each.

Attached is a log file and the updated Mechanum file and a log file from a Localization Test.

### Tuning Files

`package org.firstinspires.ftc.teamcode;

import androidx.annotation.NonNull;

import com.acmerobotics.dashboard.canvas.Canvas;
import com.acmerobotics.dashboard.config.Config;
import com.acmerobotics.dashboard.telemetry.TelemetryPacket;
import com.acmerobotics.roadrunner.*;
import com.acmerobotics.roadrunner.AngularVelConstraint;
import com.acmerobotics.roadrunner.DualNum;
import com.acmerobotics.roadrunner.HolonomicController;
import com.acmerobotics.roadrunner.MecanumKinematics;
import com.acmerobotics.roadrunner.MinVelConstraint;
import com.acmerobotics.roadrunner.MotorFeedforward;
import com.acmerobotics.roadrunner.Pose2d;
import com.acmerobotics.roadrunner.Pose2dDual;
import com.acmerobotics.roadrunner.ProfileAccelConstraint;
import com.acmerobotics.roadrunner.Time;
import com.acmerobotics.roadrunner.TimeTrajectory;
import com.acmerobotics.roadrunner.TimeTurn;
import com.acmerobotics.roadrunner.TrajectoryActionBuilder;
import com.acmerobotics.roadrunner.TurnConstraints;
import com.acmerobotics.roadrunner.VelConstraint;
import com.acmerobotics.roadrunner.ftc.DownsampledWriter;
import com.acmerobotics.roadrunner.ftc.Encoder;
import com.acmerobotics.roadrunner.ftc.FlightRecorder;
import com.acmerobotics.roadrunner.ftc.LazyHardwareMapImu;
import com.acmerobotics.roadrunner.ftc.LazyImu;
import com.acmerobotics.roadrunner.ftc.LynxFirmware;
import com.acmerobotics.roadrunner.ftc.OverflowEncoder;
import com.acmerobotics.roadrunner.ftc.PositionVelocityPair;
import com.acmerobotics.roadrunner.ftc.RawEncoder;
import com.qualcomm.hardware.lynx.LynxModule;
import com.qualcomm.hardware.rev.RevHubOrientationOnRobot;
import com.qualcomm.robotcore.hardware.DcMotor;
import com.qualcomm.robotcore.hardware.DcMotorSimple;
import com.qualcomm.robotcore.hardware.DcMotorEx;
import com.qualcomm.robotcore.hardware.HardwareMap;
import com.qualcomm.robotcore.hardware.IMU;
import com.qualcomm.robotcore.hardware.VoltageSensor;

import org.firstinspires.ftc.robotcore.external.navigation.AngleUnit;
import org.firstinspires.ftc.robotcore.external.navigation.YawPitchRollAngles;
import org.firstinspires.ftc.teamcode.messages.DriveCommandMessage;
import org.firstinspires.ftc.teamcode.messages.MecanumCommandMessage;
import org.firstinspires.ftc.teamcode.messages.MecanumLocalizerInputsMessage;
import org.firstinspires.ftc.teamcode.messages.PoseMessage;

import java.lang.Math;
import java.util.Arrays;
import java.util.LinkedList;
import java.util.List;
[... 424 lines cut ...]
        c.setStroke("#3F51B5");
        c.strokePolyline(xPoints, yPoints);
    }

    public TrajectoryActionBuilder actionBuilder(Pose2d beginPose) {
        return new TrajectoryActionBuilder(
                TurnAction::new,
                FollowTrajectoryAction::new,
                new TrajectoryBuilderParams(
                        1e-6,
                        new ProfileParams(
                                0.25, 0.1, 1e-2
                        )
                ),
                beginPose, 0.0,
                defaultTurnConstraints,
                defaultVelConstraint, defaultAccelConstraint
        );
    }
}
`

### Robot Logs

[2025_07_27__18_05_58_107__LocalizationTest.log](https://github.com/user-attachments/files/21458170/2025_07_27__18_05_58_107__LocalizationTest.log)

maheshshan, 2025-07-28:
Just wanted to add my experience too. Seems issues may be same. 
1)  During the LocalizationTest X,Y is not updating, Heading is working fine
2) in the ManualFeedForwardTuner, vref is working but robot is not moving
3) Push test worked fine, shows number of ticks, deadwheel debugger also works fine.

maheshshan, 2025-08-01:
Seems GoBildaPinpointDriver needs a fix. I have tested the following 

driver.getPosX & driver.getPosX these variables giving zeros. and 

setEncoderResolution is not accepting inPerTick

for time being do the following, it works:

convert inPerTick to mmPerTick, update the following in PinpointLocalizer

        double mmPerTick = inPerTick *25.4;
        driver.setEncoderResolution(1 / mmPerTick, DistanceUnit.MM); // Not Accepting inPerTick. Use mmPerTick
        driver.setOffsets(mmPerTick * PARAMS.parYTicks, mmPerTick * PARAMS.perpXTicks, DistanceUnit.MM);

Replace following function in PinpointLocalizer since driver.getPosX & Y gives zero values:

    @Override
    public PoseVelocity2d update() {
        driver.update();
        Pose2D pose2D = driver.getPosition();
        if (Objects.requireNonNull(driver.getDeviceStatus()) == GoBildaPinpointDriver.DeviceStatus.READY) {
            txPinpointRobot = new Pose2d(pose2D.getX(DistanceUnit.INCH), pose2D.getY(DistanceUnit.INCH), driver.getHeading(UnnormalizedAngleUnit.RADIANS)); //pose2D.getHeading(AngleUnit.RADIANS)
            Vector2d worldVelocity = new Vector2d(driver.getVelX(DistanceUnit.INCH), driver.getVelY(DistanceUnit.INCH));
            Vector2d robotVelocity = Rotation2d.fromDouble(-txPinpointRobot.heading.log()).times(worldVelocity);

            return new PoseVelocity2d(robotVelocity, driver.getHeadingVelocity(UnnormalizedAngleUnit.RADIANS));
        }
        return new PoseVelocity2d(new Vector2d(0, 0), 0);
    }

PranavDharanipathy, 2025-08-08:
For me, even with kA at 0, the bot is overshooting, the blue line goes above the red and the PID is fine except for forward-backward, the axialGain being 1 makes the bot overshoot like crazy and ram into the field barrier, how to would I fix this? Cuz it's just 1 gain...

PranavDharanipathy, 2025-08-10:
for public PoseVelocity2d update() {} function, for me if I just kept it the same:

@Override
    public PoseVelocity2d update() {
        driver.update();
        if (Objects.requireNonNull(driver.getDeviceStatus()) == GoBildaPinpointDriver.DeviceStatus.READY) {
            txPinpointRobot = new Pose2d(driver.getPosX(DistanceUnit.INCH), driver.getPosY(DistanceUnit.INCH), driver.getHeading(UnnormalizedAngleUnit.RADIANS));
            Vector2d worldVelocity = new Vector2d(driver.getVelX(DistanceUnit.INCH), driver.getVelY(DistanceUnit.INCH));
            Vector2d robotVelocity = Rotation2d.fromDouble(-txPinpointRobot.heading.log()).times(worldVelocity);

            return new PoseVelocity2d(robotVelocity, driver.getHeadingVelocity(UnnormalizedAngleUnit.RADIANS));
        }
        return new PoseVelocity2d(new Vector2d(0, 0), 0);
    }
 and it worked but when I converted the DistanceUnit.INCH to DistanceUnit.MM it when 25.4 times the distance and the turn radius was also multiplied by 25.4.