acmerobotics/road-runner-quickstart #509: How to Use and Cancel Trajectory

PranavDharanipathy, 2025-08-16:
Hello,

So I'm using `new Trajectory()` as an input for `TimeTrajectory()` but I don't know what parameters to give to the `Trajectory()` or even how to create a trajectory just as variable. Could anyone please help me with this.

rbrott, 2025-08-17:
There are higher-level APIs for creating trajectories like `TrajectoryActionBuilder` that I recommend you use. Check out this auto tutorial for some examples: https://rr.brott.dev/docs/v1-0/guides/centerstage-auto/.

PranavDharanipathy, 2025-08-17:
But the `CancelableTrajectoryAction` you guys have in the Cancellation doc needs an input of `TimeTrajectory()` which accepts 'new Trajectory()`, not `TrajectoryActionBuilder`